dmherrington/ECM-Control-System #47: Lost Connection to Encoder?

mbromley-voxel, 2018-10-05:
Motor was still controllable, but the plot for the encoder was not changing position and the software showed it was not connected to galil. Currently not sure of the cause, but it seems that the plot itself is freezing. After closing the software, it will not re-open without rebooting.

mbromley-voxel, 2018-10-05:
@majorpr13 after executing a move to home command the plot screen freezes until another move is made

mbromley-voxel, 2018-10-10:
The graph doesn't freeze, but could be confusing to the user what the current state of the machine is.  If I jog the motor down, manually reset the home position (DP*=0) then move the motor up slightly.  The display on the front panel for the current position reads 429496389 while if I submit a TP command to Galil I get -3398. 

It would be helpful to display a message in the "Current Position (um)" field when in this null/negative state such as "Reset home position".

mbromley-voxel, 2018-10-10:
In addition if you jog the motor to the home position it can overshoot the limit at times and report a negative value.  This wrecks the graph and ruins the scaling on the y-axis as it reports the current position as a similar crazy large number reported in the last comment.

majorpr13, 2019-04-26:
I dont believe this is an issue any longer with new software releases supporting ACS.

closed 2019-04-26 by majorpr13