dmherrington/ECM-Control-System #51: Software closed during run

mbromley-voxel, 2018-10-11:
With no interaction from the user, the software window closed while  a run was in progress. Re-opening the software dropped the motor. It may take some time to diagnose why it closed, but can we add some fail-safes during the programs opening routine to leave the motor in place if it is already on?

majorpr13, 2019-04-26:
I believe this issue is no longer pertinent.

closed 2019-04-26 by majorpr13

mbromley-voxel, 2019-04-26:
@majorpr13 is it possible that the software can crash and leave the motor on? If it does will the motor be disabled upon reopening the software?

majorpr13, 2019-04-26:
When the software crashes it will leave the motor on in whatever state it was last. Upon reopening the software, yes it is going to disable it and setup the machine again. Did you see it crash?