dmherrington/ECM-Control-System #54: Software Closed During Touch-off routine

mbromley-voxel, 2018-10-31:
Order of events:
1. new galil script was loaded
2. motor turned on and advanced toward the part
3. touch-off begin point established. 
4. Touch-ff routine started
5. "clear all data" function used to clear graph and the software closed immediately. The software was left on overnight, so a large amount of data was on the plot, so that may have something to do with it.

mbromley-voxel, 2018-10-31:
Software was re-opened and all above action items completed without failure.

mbromley-voxel, 2018-10-31:
"Clear all data" button just caused the software to close again. Data had not been cleared in ~ 1.5 hours

mbromley-voxel, 2018-12-20:
This continues to be an issue.  It happened again today.

The software was loaded, motor enabled, homed, and a touch off executed from the touch off utility.  While the touch off was running the "clear all" was selected to clear the graph and the software shut down.

majorpr13, 2018-12-20:
We probably have increased the chances of having this error with a higher update rate. It is obviously a thread and destroy issue that I can look into.

majorpr13, 2019-04-26:
I believe this has been addressed and therefore closing the issue.

closed 2019-04-26 by majorpr13