dmherrington/ECM-Control-System #64: Status bar messages

mbromley-voxel, 2018-12-18:
Status bar messages to display critical states/errors.  This includes munk error, touch off errors, etc.

majorpr13, 2019-01-02:
Messages now have access to the message bar. They are only displayed for a 3 second interval and are overwritten potentially faster if new messages arrive while the current one is being displayed. @mbromley-voxel is this sufficient to close this enhancement? It would appear there is another request to have a rolling history of status messages that perhaps are timestamped that could offer the operator more descriptive output.

mbromley-voxel, 2019-01-02:
Yes we will take care of the time stamped history with issue #70

majorpr13, 2019-04-29:
Closing this per the comment above.

closed 2019-04-29 by majorpr13