dmherrington/ECM-Control-System #68: Touchoff Utility use values displayed when executing

mbromley-voxel, 2018-12-20:
If the user does not click on "Set" or enter after adjusting values then the change is not recorded.  The values displayed to the user in the two data entry fields should always be used when execute of the touch off is selected.  

Update these fields to record the value on change whether or not the set button/enter is hit by the user.

majorpr13, 2018-12-20:
This function was disconnected when the new multiop interface was created as the functionality required a slightly different interface. This can be reconnected whenever we are ready to test.

mbromley-voxel, 2018-12-20:
Will reconnecting effect any of the performance of the multiop touch off routine?

majorpr13, 2018-12-20:
No it should not.