dmherrington/ECM-Control-System #70: Running log of status bar messages

mbromley-voxel, 2018-12-20:
Allow the user to click into the status bar to view a history of the status messages displayed.

majorpr13, 2018-12-20:
Instead of a status bar opening, can we have a window that is expandable from the primary interface that displays a running log, appropriately timestamped and color coordinated?

mbromley-voxel, 2018-12-20:
Just what we need... another window...

majorpr13, 2018-12-20:
It would be apart of the main window...still a showable box...we just dont have more vertical room.

majorpr13, 2019-08-05:
This is now integrated within the main GUI. Color coding will be a feature request, however, the ACS messages from a profile will have no color coordinating as there is no way to understand the level of severity of the message.

closed 2019-08-05 by majorpr13