dmherrington/ECM-Control-System #74: Software closes on "run configuration' call

mbromley-voxel, 2019-04-22:
Happened two more times today (4/22/19)

majorpr13, 2019-04-24:
Right so this is what i was expecting. It still believes the machine is moving. You should be able to hit stop to clear it out of this state. I can work on a fix that checks either for motion and/or transmits a stop command before a motion is run.

mbromley-voxel, 2019-04-24:
hitting the stop button clears the state and lets machining continue

mbromley-voxel, 2019-04-24:
Would it be possible to add an error pop up if the state is not as needed/expected before executing a call so that the software doesn't just close?  We can work on making it more robust, but there still may be cases where we get it stuck in a state and it would be ideal to not just shut down.

majorpr13, 2019-04-24:
yeah I can block the close conditions, useful in debug mode as they get caught. I can add notifications that precheck the state before advancing and not shut down. But what you want to do to handle each state is dependent upon the state and the subsequent action you are about to take. We can work out details later but I would need a flow diagram from each state, action, etc on how to handle and mitigate.

majorpr13, 2019-04-26:
This should no loner be an issue as it blocks the user from doing things in unintended states that would cause this.

closed 2019-04-26 by majorpr13