dmherrington/ECM-Control-System #76: No Notification of Error when Power Supply is Off

mbromley-voxel, 2019-04-23:
The software knows to kick out of the run if it does not have a connection to the power supply, however, it does not notify the user of why it stopped the run, it just returns to the idle state, leading the user to believe there is a larger issue.

mbromley-voxel, 2019-04-30:
@majorpr13 Is this now validated?

majorpr13, 2019-04-30:
software is in there to detect, how the user is notified is indirect. The communications light will not turn green and there will be no alert in the notification tab that the munk is connected. In my current code I have a notification message for when not connected that will be pushed up tonight. So the user still has to be cognizant of their display. If you want further in the operator face I can do that as well.

mbromley-voxel, 2019-04-30:
Understood the light is fine until the operator attempts to do something with the ACS.  At that time the software should alert the user that Munk is not connected or comms do not exist.  Some generic message related to comms.

majorpr13, 2020-01-30:
This has been addressed in the current branch of code.

closed 2020-01-30 by majorpr13