dmherrington/ECM-Control-System #77: stop button does not stop motion profile when running routine

mbromley-voxel, 2019-04-24:
The stop button needs to execute the sequence "stopall;killall;OUT0.2 = 0; start 2, SET_DEFAULTS_GOTO"

majorpr13, 2019-04-24:
Ok I need a little more clarity. Currently it calls killAll which should be stopping all motion. Does this not also terminate the program executing in the buffer? I do not have access to a stopAll command off the C++ interface, I have a int _ACSCLIB_ WINAPI acsc_StopBuffer(HANDLE Handle, int Buffer, ACSC_WAITBLOCK* Wait); which will stop whatever code executing in a buffer.....is this what you are looking for? I need clarity on the later commands...what are you trying to do with them? what do they need to do? is the SET_DEFAULTS_GOTO a label to be executing? What is OUT0.2 = 0? Is this a DIO pin you are turning off the power supply? What is start 2 doing? 

Also, do you want this function to always occur on the stop button?

mbromley-voxel, 2019-04-24:
1. It only stops the current motion, a vibration profile is a string of motions, so, it just proceeds to the next one. Stop all kills the execution of the script so it does not trigger the next motion.
2. It does not kill the buffer.
3. Yes, acsc_StopBuffer is what you need to call. if you can't tell it to stop all of the buffers, then you will have to set it up to run through every accessible buffer.
4. "start 2, SET_DEFAULTS_GOTO" is a single command. If you kill the execution of the buffers, it will not reset the PID parameters for the linear motor, which, are different for simple motions (jog, move, etc.) and complex motions (vibrations).
5. OUT0.2 = 0, sets the power supply output to 0, killing the power supply. How were you killing the power supply before?
6. In recap, we need to 1) kill the buffers, 2) kill the motion, 3) kill the power supply, 4) reset the PID values to default, 5) change operation status to complete 6) whatever else the software needs to return to normal functionality.

Yes, this should be done with every push of the stop button.

majorpr13, 2019-04-24:
4. I can see what I can do with that.
5. OUT0.2 is the format of that port 0 bit 2

majorpr13, 2019-04-24:
This should be addressed via https://github.com/mbromley-voxel/ECM-Control-System/commit/e740bb63e9e326cd69af08e3a78804fae31e0819.

closed 2019-04-26 by majorpr13