dmherrington/ECM-Control-System #90: Profile Configuration -> Operations

rhowesVI, 2020-02-14:
If I add a new operation to a profile configuration nothing happens when attempting to run the operation. It acts similar to if there was no connection to the power supply.

Using an operation within a previously saved profile configuration is possible, and it is how I've worked around the issue. I can edit old operations, but adding an operation to a saved profile configuration also doesn't work. 

Also, opening some previous profile configurations causes the program to crash.

rhowesVI, 2020-02-14:
I wasn't able to replicate the issue.

majorpr13, 2020-03-18:
This issue seemed to have been solved as a result of the linking the munk power supply with the default state of the mode in the profile configurations. It appears it was always starting as dead, however, having the display in a different state. The original find was by changing the state actually then would sync the display and underlying data. This has now been fixed.

closed 2020-03-18 by majorpr13